furrtek/portapack-havoc #238: Importing c16 recording files into universal radio hacker

ril3y, 2019-04-23:
I am not sure this is the right place for the question / comment but here it goes.  I have recoreded via the portapack some signals I want to process on Universal Radio Hacker however it seems that they are overly large and just do not work.  Using the hackrf in hackrf mode with a pc and URH to capture the same signals work just fine.  

I am I doing something wrong when capturing?  Should / can I use a different format?  What is  c16 also.  I was unable to find much documentation on this.

Thanks,

robife, 2019-05-03:
You can open c16 file, with software audacity.
There you can edit you catched signal.

furrtek, 2019-08-07:
The format info was added here: https://github.com/furrtek/portapack-havoc/wiki/Capture
URH might be expecting values with reverse endianness.