furrtek/portapack-havoc #244: Possibility to generate custom bmp_splash.hpp

BorjanEch0, 2019-06-21:
Hi!
I am trying to generate a custom splash image, i located the splash file and i also found the python tool for bitmap generation, i tried it but it gave a different type of file, i tried to copy only the bytes and calculate the length and enter it in the variable but the splash showed blank. what am i missing, the image used was 128x128 png, any help would be appreciated.
Thanks in advance.

specht, 2019-06-21:
You need a 16 color BMP file, this worked for me.

furrtek, 2019-06-21:
The Python tool is only used for icons (1BPP bitmaps). For the splash image you only need to copy data from a 4BPP bitmap as @specht wrote. You may need to alter the code to center it.

BorjanEch0, 2019-06-25:
Excellent! Thank you!

closed 2019-06-25 by BorjanEch0

AfanasyPopov, 2019-11-28:
> The Python tool is only used for icons (1BPP bitmaps). For the splash image you only need to copy data from a 4BPP bitmap as @specht wrote. You may need to alter the code to center it.

Can you share info about image converter to 4BPP bitmap?
I try to use:
bmp-converter
bmpsuite
Image2Lcd
ImageKit9eTrial
lcd-image-converter
Bitmap2Code
winhex

no chance to make 16 color & 4BPP & RLE compressed bitmap (
Please help with this problem ?