furrtek/portapack-havoc #245: unable to build latest

imzenreally, 2019-06-29:
i just picked up a hackrf and portapack - and put the bin file that is in firmware/bin on the portapac and it works fine.

but i wanted to build the latest to ensure i had all the latest features etc - but no matter what i try i am unable to get the current package to build.  does anyone have any suggestions?  or know of a place where there are monthly builds or something?

ImDroided, 2019-06-29:
Post the error you are getting or we won't be able to help you out.

imzenreally, 2019-06-29:
64475 :: ~ » cmake --version                                                                                  
cmake version 3.5.1                                                                                                                                                                                                                             CMake suite maintained and supported by Kitware (kitware.com/cmake).   
---------------
64475 :: ~ » lsb_release -a                                                                                    
No LSB modules are available.                                                                                           
Distributor ID: Ubuntu                                                                                                  
Description:    Ubuntu 16.04.6 LTS                                                                                      
Release:        16.04                                                                                                   
Codename:       xenial                                                                                                  
64475 :: ~ »   
---------------------------------
dfu-util 0.8 
---------------------------------
[ 25%] Built target baseband_shared
[ 25%] Built target baseband_siggen.elf
[ 25%] Performing update step for 'hackrf'
Current branch master is up to date.
[ 26%] Performing configure step for 'hackrf'
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr/arm-none-eabi
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tylerv/portapack-havoc/build/firmware/hackrf-prefix/src/hackrf-build
[ 26%] Performing build step for 'hackrf'
[  3%] Performing build step for 'libopencm3_hackrf_usb'
  GENHDR  include/libopencm3/efm32/efm32g/irq.yaml
Makefile:58: recipe for target 'include/libopencm3/efm32/efm32g/irq.yaml.genhdr' failed
hackrf_usb/CMakeFiles/libopencm3_hackrf_usb.dir/build.make:110: recipe for target 'hackrf_usb/libopencm3_hackrf_usb-prefix/src/libopencm3_hackrf_usb-stamp/libopencm3_hackrf_usb-build' failed
CMakeFiles/Makefile2:949: recipe for target 'hackrf_usb/CMakeFiles/libopencm3_hackrf_usb.dir/all' failed
CMakeFiles/Makefile2:674: recipe for target 'hackrf_usb/CMakeFiles/hackrf_usb.dfu.dir/rule' failed
Makefile:300: recipe for target 'hackrf_usb.dfu' failed
firmware/CMakeFiles/hackrf.dir/build.make:110: recipe for target 'firmware/hackrf-prefix/src/hackrf-stamp/hackrf-build' failed
CMakeFiles/Makefile2:93: recipe for target 'firmware/CMakeFiles/hackrf.dir/all' failed
CMakeFiles/Makefile2:138: recipe for target 'firmware/CMakeFiles/firmware.dir/rule' failed
Makefile:131: recipe for target 'firmware' failed

imzenreally, 2019-06-29:
i'm using the ppa from ubuntu - i also downloaded and set up the gcc-arm update stuck it in usr/local/ chmod'ed it etc and then copy pasted the config in the wiki and tried that as well to the same results.

imzenreally, 2019-06-29:
i'm building on linux subsystem for windows - i've also tried building on my 18.04 ubuntu laptop (which is the machine i use to talk to my hackrf)

uglurass, 2019-08-02:
same here tyler. seems python says no such file or directory.

TheBizzotech, 2019-08-10:
Hey there,

I was ableto compiling the last firmware with my Ubuntu 18.04.3 LTS, below the commands that I hit:

`cd /opt` this is not necessary it's my convention to operate in that directory...
`sudo -s`
`apt install gcc-arm-embedded gcc-arm-none-eabi-source linux-libc-dev-armel-cross linux-libc-armel-cross binutils-arm-linux-gnueabi libncurses5-dev dfu-util cmake git pyhton2.7`
`wget https://bootstrap.pypa.io/get-pip.py`
`python2.7 get-pip.py`
`pip install pyyaml`
`git clone https://github.com/furrtek/portapack-havoc.git`
`cd portapack-havoc`
`mkdir build`
`cd build`
`cmake ..`
`make firmware`

The final results:

> [100%] Linking CXX executable application.elf
[100%] Built target application.elf
Scanning dependencies of target application
[100%] Generating application.bin
[100%] Built target application
Scanning dependencies of target firmware
[100%] Generating portapack-h1-havoc.bin
[100%] Built target firmware

hope this could help.

imzenreally, 2019-08-10:
woop - this works - thanks for your help - much appreciated

TheBizzotech, 2019-08-11:
> woop - this works - thanks for your help - much appreciated
> 
> looks like the wiki has been updated as well to reflect most of the above commands -- thanks again - built a new image and flashed my portapack with no issues whatsover
> 
> thanks

Delighted I could help :)

closed 2019-10-14 by furrtek