furrtek/portapack-havoc #246: C16 file format complex, float...?

VeloVolo, 2019-07-01:
What data contain these files? complex, float..?
Could not find any information about these files.
May be someone could advise how to open these files in GNU Radio?

furrtek, 2019-07-01:
C16 means Complex 16-bit. It's a stream of 16-bit integer pairs for I and Q.
Try the "ishort to complex" block in GRC.

VeloVolo, 2019-07-01:
Thank You! That is i was looking for!

TeleportingKitten, 2019-08-04:
Can I decrypt the C16 file format?  Man code is definitely what I lack in. I put the file in a hex edit but it shows nonsense because it's encrypted.

furrtek, 2019-08-04:
It's not encrypted. Why would it be ?
You're seeing pairs of signed 16-bit values as stated above.

furrtek, 2019-08-07:
Info added to https://github.com/furrtek/portapack-havoc/wiki/Capture

closed 2019-08-07 by furrtek