furrtek/portapack-havoc #251: SDRAngel incompatibility - build on newer firmware?

Sinager, 2019-07-28:
Hi,
I'm running a very early HackRf with a very early green PortaPack.

I'd love to be able and use it with SDRAngel in HackRf mode, but looks like the Havoc "HackRf mode" isn't compatible, with the device not being recognized.

With freshy built bins I get

_Serial number: RunningFromRAM
Board ID Number: 2 (HackRF One)
Firmware Version: git-ef5b907 (API:1.03)
Part ID Number: 0xa000cb3c 0x00484748_

not working with SDRAngel

but HackRf alone 
_Serial number: 000000000000000014d463dc2f9b9be1
Board ID Number: 2 (HackRF One)
Firmware Version: git-509c8f4 (API:1.03)
Part ID Number: 0xa000cb3c 0x00484748_
works fine.

Is there a common man fix?

Cheers
Stefano IZ0MJE

jboone, 2019-07-28:
Hi Stefano. I think this is due to the serial number appearing as "RunningFromRAM", which I suspect SDRAngel does not like. I *thought* this issue was addressed months ago. I will check on this on my [sharebrained/portapack-hackrf](https://github.com/sharebrained/portapack-hackrf) project when I'm at work tomorrow. If this bug still exists, it would be there.

Sinager, 2019-07-28:
Hi,
thanks for looking into the matter.
I had a suspicion in that direction, but unfortunately nothing as simple as grepping and replacing that string with a number.
Stefano

Sinager, 2019-08-04:
> I _thought_ this issue was addressed months ago. 

Hi Jared,
I confirm it's totally fine when running the stock PortaPack firmware, which indeed returns a correct serial number.
It's just Havoc not working: maybe it beacause not building on your latest.

Cheers
Stefano

jboone, 2019-08-04:
Hi Stefano. Sorry for the delay.

Can you tell me exactly which version of HAVOC you're using? Are you using a pre-built binary, or are you building it yourself?

jboone, 2019-08-04:
Oh, sorry, you said "freshly-built". Investigating...

jboone, 2019-08-04:
I can confirm there are some changes in portapack-hackrf that are not in this repo that *could* impact the serial number. I'm getting my brain back into "git" mode to do a merge on my fork of the HAVOC repo, which will appear here: https://github.com/jboone/portapack-havoc/

Sinager, 2019-08-05:
Brilliant, thanks!

jboone, 2019-08-12:
@Sinager My apologies for the delay. I've been traveling and haven't had time to work on this... until now. Please check my fork, here, and let me know if it fixes the issue:

https://github.com/jboone/portapack-havoc/

If so, I'll create a pull request for Furrtek to bring the project repo up to date with my PortaPack repo changes.

jboone, 2019-08-12:
@Sinager Also, I have "nightly" builds appearing here, if you want to try that or otherwise don't want to build it for yourself.

https://jboone.github.io/portapack-havoc-nightly/

scorpionul222, 2019-08-13:
Hello the last version presented above in the link is blocked when I select the capture mode
sdrangel works fine now

Sinager, 2019-08-13:
Hello Jared,
cool, it seems to work fine.
Capture / replay seems ok too.

Thank you very much for the support!
All the best
Stefano

scorpionul222, 2019-08-14:
freezes at the menu capture 
Firmware 
https://jboone.github.io/portapack-havoc-nightly/
Solution?

bicurico, 2019-09-07:
Hi,

I confirm that the MASTER branch of Havoc is using GIT VERSION as serial number instead of SYSTEM VERSION. This causes SDRAngel to not recognize the HackRF One with Portapack running with Havoc.

The solution presented in this thread works fine - SDRAngel will work.

I have not noticed any issue with CAPTURE MENU. I captured and replayed successfully.

Regards,
Vitor

aneitzel, 2019-11-17:
latest build "PortaPack-HAVOC-2019-09-26-6e941ea.tar.xz " from jboone has issue with freeze after select capturing :-( 

Latest working is PortaPack-HAVOC-2019-09-23-cab22ed.tar.xz	

HackRF + Portapack (blue one from 2018 10 29)

Seems to be a issue in the build