furrtek/portapack-havoc #255: error no file system??

TeleportingKitten, 2019-08-01:
I'm getting an error when I click the little red R icon when using the Capture, it's saying error at the top left and telling me I have no File system? I click OK and it brings me back to the Hack RF one capturing screen. It's definitely capturing because I see the spectrum, but am I missing files? I have a 64gb SD card inserted.

furrtek, 2019-08-01:
Is the card FAT32 formatted ? Is the capture file present and >0 in size ? Maybe a block size problem.

TeleportingKitten, 2019-08-01:
I was actually trying to delete the post if I could haha, I'm new to github, I got it working. Basically it needs to be a FAT format or FAT32. I have another problem though, I'm not able to generate the world_map.bin file. Do I just dubble click on the python adsb_map.py? When I do it just opens and closes extremely fast. Do I need a certain python version?

furrtek, 2019-08-01:
Run it from a terminal, it's probably giving you an error but you can't see it. `python adsb_map.py`.
Python 2.7 is enough, might be broken with Python 3.

TeleportingKitten, 2019-08-01:
I'm using windows 10, and I've tried python 3 and 2.7 I'm installing kali Linux to see if will be able to build from a Linux system instead of a windows. All let you know if it works.

furrtek, 2019-08-01:
You shouldn't need Linux for that. Just run python from a cmd prompt and see what it tells you :)

TeleportingKitten, 2019-08-04:
I'm not sure how to run the file from windows python correctly so I'm using Linux. So I built the map file... but I think the map file is somewhat broken because it shows only 3 locations on the map where the plane is, and the map is half cut from the screen? Maybe my SD card is to old to even run such a big file in real time. But all post a photo here if I can.

furrtek, 2019-08-07:
I doubt a 64GB (so certainly not too old) would be the cause.
A photo or any error/warning messages would help.

closed 2019-10-14 by furrtek

Artorios87, 2020-01-21:
продавец сказал что поддерживает 128gb!

Artorios87, 2020-01-21:
отформатировал в fat-32  Thanks

Jean-france, 2020-02-01:
Hi all,
I have the same problem ; I can't Capture, the hack write a error : **no filesystem**.
I have the Havoc firmware and i wipe my 64 Go SD card, but nothing.
Coulg you help, if you have any idea...
Thanks in advance.
Best regards,
Jean

Ahorrorfilm, 2020-02-01:
Be sure you put the files from "sdcard" on your..sdcard, but if you've
already did that then you probably have the splash screen issue..so disable
the splash on startup or whatever it's called (sorry I have horrid memory
and currently dont have my hackrf on me) then turn it off and back on and
your files should be there. Also, that was an issue I experienced with
older firmware and haven't experienced it anymore since the newer releases,
so maybe even flashing the latest release would help. Hopefully that helps,
sorry if that's all to ran together...I know sometimes there's a language
barrier here..

Jean-france, 2020-02-01:
Many thanks for your reply.
"Be sure you put the files from "sdcard" on your..sdcard, ", ? How you do this ? My sdcard is empty, nothing inside....

furrtek, 2020-02-01:
https://github.com/sharebrained/portapack-hackrf/wiki/General-Use
Format to FAT32, not ExFAT. Copy/paste "sdcard" folder to the root of the sd card.