furrtek/portapack-havoc #257: OOK Transmit menu freeze

Sceeker, 2019-08-12:
In the OOK TX menu, when I try to scroll past the **12E** type, everything freezes and I have to restart.
I didn't try to fix the bug yet so I can't give much more details for now.

z4ziggy, 2019-11-05:
Fixed here: https://github.com/z4ziggy/portapack-havoc/commit/39a6c34cfe1a81b9fa90918c8ccc5798badd7131

I tried a pull req, but github is crazy and my best fix is reinit my repo.

Sceeker, 2019-11-19:
> Fixed here: [z4ziggy@39a6c34](https://github.com/z4ziggy/portapack-havoc/commit/39a6c34cfe1a81b9fa90918c8ccc5798badd7131)
> 
> I tried a pull req, but github is crazy and my best fix is reinit my repo.

Just recompiled it (couldn't try it before, work takes me so much time :/ ) and now it verks! Thanks :)

closed 2019-11-19 by Sceeker