furrtek/portapack-havoc #259: portapack amplifier

moayadassil, 2019-08-23:
I realized a transmitter based GNU radio and I can amplify the signal up to 3w using an amplifier. but with the portapack I can not. I tried to analyze the source code of the firmware havoc and I found that the gain of tx is 47 dbm even the gain of my transmitter gnu radio. what i have to do to amplify my signal portapack

furrtek, 2019-08-23:
> I realized a transmitter based GNU radio and I can amplify the signal up to 3w using an amplifier.

I don't get it. What SDR did you use for that ? Can't you use the same amplifier with the Portapack ?

> I found that the gain of tx is 47 dbm even the gain of my transmitter gnu radio.

That's 47dB, it's the VGA range. It would be worrying if the HackRF could output 50W.
GNU Radio can't "create power", the max transmit power is still defined by your hardware.
If you "amplify" the data in GRC then it'll saturate and the signal will be distorted (but not more powerful).

> what i have to do to amplify my signal portapack

Use an rf amp :) A real one. There are tons of them of wildly varying quality on that-Chinese-website and that-american-auction-website. It goes without saying that you should be careful where you blast the watts and if you have the right to.

moayadassil, 2019-08-23:
ok thank you Sir