furrtek/portapack-havoc #261: Play Dead mode

ArcanA998, 2019-08-31:
Hi, I would like to know when the login mode will be available with play dead, I think it is very useful and indispensable for this device and I would really need it, thanks in advance for the answer.

bicurico, 2019-09-13:
This is currently deactivated in the sources, so I activated it again. The result was not so good: while the login now was mandatory, after entering the pass code, the device would just show PRESS RESET.

The problem is that after a RESET it would just show the fake error message again: I was in a loop.

I had to actually upload the previous firmware in DFU mode to unbrick my HackRF with Portapack, because I could not enter the regular HackRF mode to flash any firmware.

To fix the play dead with login function, it seems that the whole play dead and login functionality needs to be improved and I think that would require three states to be written in battery backed up RAM: State 1 = no login required, State 2 = login required, State 3 = one time no login

State 1: never changes by itself
State 2: changes to 3 if pass code was entered successfully
State 3: changes to 2 if login is active in Settings Menu otherwise changes to 1

Activate Play Dead mode in menu will activate State 2 and lock right away.
Activate Login in Settings Menu will activate State 2 but not lock right away - only after RESET.

I am not sure that I am literate in C++ enough to get this done, but I will try.

The better option would be to edit the play dead routine to not require a RESET after the pass code was entered successfully. It should just exist the play dead app and return to the main menu.

Regards,
Vitor

bicurico, 2019-09-14:
Hi,

I managed to get it working in a satisfactory way to me.

Download here: https://www.mediafire.com/file/kpi4a550iux9fhk/portapack-h1-havoc-vma.bin/file

WARNING: Make sure you have a backup battery (Lithium battery cell) and you SET A PASS CODE before flashing this firmware. If you don't, you may get trapped in a loop, because you don't know the PASS CODE and the default one may nor work (RLDU).

What this does:

1) You activate LOGIN on RESET in the SETTINGS
2) When you now RESET the device, it will play dead until you enter the PASS CODE
3) If you activate PLAY DEAD from the MENU, it will play day until you enter the PASS CODE, even if you RESET it

If this breaks and you end up in a loop (after RESET you see the PLAY DEAD and the PASS CODE does not work, you will need to reflash in DFU mode - no other way).

Hope this helps.

Regards,
Vitor

PS: I am not uploading th emodified sources, because I don't know how to use Github... If someone wants the modified sources, just pop me a message and/or visit my blog at https://vma-satellite.blogspot.com/

ArcanA998, 2019-09-16:
Hi, first of all I thank you for the time you've spent with us.

I will try to flash it soon, but first a question: Was this firmware created on the latest version of the Havoc? It is up to date ? Thanks again you're a life saver!

bicurico, 2019-09-16:
Hi,

Yes, I think I used the latest version – at least I think I did, as I am not literate with Github (still learning).

Meanwhile I updated a few more things, so I am not sure you have the latest version, which includes:

  1.  Scanner:
     *   Shows frequency while scanning
     *   Reads the list of frequencies from a file contained in the SD root and called “scanner-freq.TXT”. This file has a very strict formatting: every frequency is expressed in Hz and needs to have exactly 10 digits and has to be edited in Windows (to add CrLF as opposed to just Lf). The last line must be “ENDENDENDE”+CrLF
     *   If the frequency is less than 142MHz, it will show “(AM)”, otherwise “(NFM)”. This is not ready, yet: my goal was to automatically switch modulation during scanning, but that does not work and crashed the device. I asked for help (still waiting) to figure out how to change modulation without crashing the device. This means that presently only NFM is supported during scan.
  2.  Play Dead:

I activated this and it kind of works:

     *   If you select Play Dead from the menu (panic), it will play dead until you enter the pass code. It will survive a reset, so unless you enter the pass code, the device is dead.
     *   If you activate Login from the Settings, it will play dead on the NEXT reboot. Again, unless you enter the pass code, the device is dead. But once you enter the pass code, you can reboot as many times you want and it won’t ask for the pass code. This is not totally correct, but was my work-around to the issue that caused the original authors to deactivate play dead in the first place.
So now you can: play dead if you are in a panic situation (a police officer is coming to ask you what you are doing…) or enable play dead once your session terminates (so you can safely put away the device, knowing nobody will mess with it).
But beware that you need to set a pass code FIRST! If you fail to do it, you will end up with a loop (= bricked device) and the only way out of it is to flash the recovery image using DFU mode.

Regards,
Vitor

Enviada: 16 September 2019 09:58
Para: furrtek/portapack-havoc <portapack-havoc@noreply.github.com>
Cc: Vitor Martins Augusto <vma@norcam.pt>; Comment <comment@noreply.github.com>

0144000000
0146000000
ENDENDENDE

ArcanA998, 2019-09-21:
Thank you Bicurico, worked for me ! With the battery inserted and the current firmware I entered the numerical sequence from the menu, after which I flashed the firmware in DFU mode (What you put on mediafire) and, with the battery inserted I went to the settings to activate the login mode. I am very happy that someone has worked on it, and although it is not perfect as a system it is enough to add an extra security system to the Portapack. THANK YOU SO MUCH !