furrtek/portapack-havoc #263: Problem At Compiling The final bin file & why does the graphic world map dosent work?

roxox666, 2019-09-01:
ive done evrything by the book and i got this ERROR:

<pre>[ 28%] <font color="#3465A4"><b>Performing configure step for &apos;hackrf&apos;</b></font>
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /
-- The C compiler identification is unknown
CMake Error at CMakeLists.txt:26 (project):
  The CMAKE_C_COMPILER:

    arm-none-eabi-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable &quot;CC&quot; or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also &quot;/home/xy/portapack-havoc/build/firmware/hackrf-prefix/src/hackrf-build/CMakeFiles/CMakeOutput.log&quot;.
See also &quot;/home/xy/portapack-havoc/build/firmware/hackrf-prefix/src/hackrf-build/CMakeFiles/CMakeError.log&quot;.
make[3]: *** [firmware/CMakeFiles/hackrf.dir/build.make:104: firmware/hackrf-prefix/src/hackrf-stamp/hackrf-configure] Error 1
make[2]: *** [CMakeFiles/Makefile2:116: firmware/CMakeFiles/hackrf.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:93: firmware/CMakeFiles/firmware.dir/rule] Error 2
make: *** [Makefile:109: firmware] Error 2
</pre>

YannickDC, 2019-09-02:
Are you sure 'arm-none-eabi-gcc' is installed on your system?

roxox666, 2019-09-02:
its apear to be a failre in instalation
 igot  an error 
xy@xy-Lenovo-IdeaPad-Y510P:~$ sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa
 This PPA is an alternative to toolchain released at https://launchpad.net/gcc-arm-embedded. The source codes for both are same. Currently supports Ubuntu 10.04/12.04/14.04/14.10 32 and 64 bit.

Detailed explanations to Launchpad PPA can be found at https://help.launchpad.net/Packaging/. That website explains how a PPA is set up and how to add existing PPA and install software from it.

Here are quick steps to install toolchain from this PPA on Ubuntu before 14.04:

Step1: Inside Ubuntu, open a terminal and input
       "sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa"

Step2: Continue to input
       "sudo apt-get update"

Step3: Continue to input to install toolchain
       "sudo apt-get install gcc-arm-embedded"

To remove installed toolchain, just input "sudo apt-get remove gnu-arm-embedded".
To update the toolchain, just repeat above step2 and step3.

If it reports error message of conflict to gcc-arm-none-eabi, which is likely if upgrading from 4.x to 5+, please uninstall it first with:
       "sudo apt-get remove gcc-arm-none-eabi"

Questions should be asked at https://answers.launchpad.net/gcc-arm-embedded

Bug can be filed at https://bugs.launchpad.net/gcc-arm-embedded/+filebug. It is highly encouraged to ask question first before filing a bug.
 More info: https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa
Press [ENTER] to continue or Ctrl-c to cancel adding it.

Hit:1 http://security.ubuntu.com/ubuntu disco-security InRelease
Ign:2 http://dl.google.com/linux/chrome/deb stable InRelease        
Ign:3 http://ppa.launchpad.net/team-gcc-arm-embedded/ppa/ubuntu disco InRelease
Hit:4 http://dl.google.com/linux/chrome/deb stable Release                    
Hit:6 http://il.archive.ubuntu.com/ubuntu disco InRelease                     
Err:7 http://ppa.launchpad.net/team-gcc-arm-embedded/ppa/ubuntu disco Release
  404  Not Found [IP: 91.189.95.83 80]
Hit:8 http://il.archive.ubuntu.com/ubuntu disco-updates InRelease
Hit:9 http://il.archive.ubuntu.com/ubuntu disco-backports InRelease
Reading package lists... Done
E: The repository 'http://ppa.launchpad.net/team-gcc-arm-embedded/ppa/ubuntu disco Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
xy@xy-Lenovo-IdeaPad-Y510P:~$ sudo apt-get update
Hit:1 http://il.archive.ubuntu.com/ubuntu disco InRelease
Hit:2 http://il.archive.ubuntu.com/ubuntu disco-updates InRelease             
Hit:3 http://il.archive.ubuntu.com/ubuntu disco-backports InRelease           
Ign:4 http://ppa.launchpad.net/team-gcc-arm-embedded/ppa/ubuntu disco InRelease
Ign:5 http://dl.google.com/linux/chrome/deb stable InRelease                  
Hit:6 http://dl.google.com/linux/chrome/deb stable Release                    
Err:7 http://ppa.launchpad.net/team-gcc-arm-embedded/ppa/ubuntu disco Release
  404  Not Found [IP: 91.189.95.83 80]
Hit:8 http://security.ubuntu.com/ubuntu disco-security InRelease
Reading package lists... Done
E: The repository 'http://ppa.launchpad.net/team-gcc-arm-embedded/ppa/ubuntu disco Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
xy@xy-Lenovo-IdeaPad-Y510P:~$ sudo apt-get install gcc-arm-embedded
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: can't find the pack  gcc-arm-embedded
xy@xy-Lenovo-IdeaPad-Y510P:~$

roxox666, 2019-09-02:
after runing 
sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa
sudo apt-get update
i had executed the cmd:
sudo apt-get install gcc-arm-embedded

i got this error:
E: can't find the pack gcc-arm-embedded

terminal view:

xy:~$ sudo apt-get install gcc-arm-embedded
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: can't find the pack gcc-arm-embedded

YannickDC, 2019-09-03:
Hey, i've just both Arch and my Ubuntu VM to build the firmware and with the instructions provided it should be working, I could email you a build firmware if you'd like? 

I have a feeling you messed up around the 'sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa' part since if you run that command you can actually see a set of instructions that could help you install it, No biggie I can still email you a build firmware. :)

v1tog, 2019-12-20:
Hi i don't know if this is the right place to put this question but i'm one day new to this site and one week with my hackrf one so here is my question can someone please point me in the right Direction on how to put files on a SD card for the portapack ie GPS simulator and world maps thank you in advance

robpegs, 2020-01-01:
Hi is it possible you could send me the latest build firmware, my computer crashed over Christmas, cheers Rob
robstyles@mail.com

closed 2020-01-01 by furrtek