furrtek/portapack-havoc #264: use git submodules

ZeroChaos-, 2019-09-02:
Things like this make it impossible to automate the build in gentoo.  Please switch to using a git submodule.
```
[  1%] Performing download step (git clone) for 'hackrf'
cd /var/tmp/portage/net-wireless/portapack-havoc-9999/work/portapack-havoc-9999_build/firmware/hackrf-prefix/src && /usr/bin/cmake -P /var/tmp/portage/net-wireless/portapack-havoc-9999/work/portapack-havoc-9999_build/firmware/hackrf-prefix/tmp/hackrf-gitclone.cmake
Cloning into 'hackrf'...
fatal: unable to access 'https://github.com/jboone/hackrf.git/': Could not resolve: github.com (Could not contact DNS servers)
Cloning into 'hackrf'...
fatal: unable to access 'https://github.com/jboone/hackrf.git/': Could not resolve: github.com (Could not contact DNS servers)
Cloning into 'hackrf'...
fatal: unable to access 'https://github.com/jboone/hackrf.git/': Could not resolve: github.com (Could not contact DNS servers)
```

jboone, 2019-09-02:
I'm pretty sure this is my doing. Sorry. When I find the time to bash my head against CMake and git, I will fix in my repo and PR to this repo.

jboone, 2019-11-16:
This was addressed in commits on September 23rd on the sharebrained/portapack-hackrf repo.