furrtek/portapack-havoc #267: How exactly do I get an SD card working?

ogdood, 2019-09-24:
I have a 128gb SD card in my portapack and Havoc is running just fine. I tried to 'Capture' a frequency and when I pressed that red circle "R", it gives me a read exclamation point inside a triangle and it reads: "no filesystem". My only option at that point is to press "OK".

I put my portapack into DFU mode and on Windows 10, I tried using diskpart to find the partition. When I did a LIST DISK, the SD card in this portapack didn't come up, just my SSD and my backup drive.

I went through the wiki and I cant find anywhere on how exactly to format an SD card, what file system its using, etc.

Any help would be really appreciated.

Im on Windows 10 but I also have a Macbook Pro and I have a Ubuntu box as well.

Thanks.

ogdood, 2019-09-24:
Please close this. I used this incredibly new and sophisticated option here on Github named the SEARCH function.

Why am I slow -_-

Solution found, thanks.

(Its FAT or FAT32)

DEFCON3187, 2019-10-13:
well? Did you resolve the issue?

furrtek, 2019-10-13:
I'm at a train station right now, those tracks are very inviting.

DEFCON3187, 2019-10-13:
LOL!!!

Lorenzo Tempany | Managing Director
SHM Projects | 3 Kings Cross Road Potts Point | Sydney  NSW  2010
m: 0404695175  | e: ltempany@shmprojects.com.au  | f: +61 2 80466293
w: www.shmprojects.com.au | abn: 68 419 031

closed 2019-10-13 by furrtek