furrtek/portapack-havoc #273: problem compiling the last firmware

F0GPI, 2019-11-07:
Hi, 
i need a solution for compiling the new firmware on Ubuntu 16.04. 

i joined my error on this post. 

[ 37%] Generating baseband_wfm_audio.bin, baseband_wfm_audio.img
make[3]: ***  Aucune règle pour fabriquer la cible « ../firmware/baseband/hackrf_usb_ram.bin », nécessaire pour « firmware/baseband/hackrf.img ». Arrêt.
CMakeFiles/Makefile2:814 : la recette pour la cible « firmware/baseband/CMakeFiles/baseband.dir/all » a échouée
make[2]: *** [firmware/baseband/CMakeFiles/baseband.dir/all] Erreur 2
CMakeFiles/Makefile2:535 : la recette pour la cible « firmware/CMakeFiles/firmware.dir/rule » a échouée
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Erreur 2
Makefile:261 : la recette pour la cible « firmware » a échouée
make: *** [firmware] Erreur 2

what the problem? 
i need help.

infodel, 2019-11-07:
I believe i am coming across the same issue. 

[ 37%] Built target baseband_mic_tx.elf
make[3]: *** No rule to make target '../firmware/baseband/hackrf_usb_ram.bin', needed by 'firmware/baseband/hackrf.img'.  Stop.
CMakeFiles/Makefile2:401: recipe for target 'firmware/baseband/CMakeFiles/baseband.dir/all' failed
make[2]: *** [firmware/baseband/CMakeFiles/baseband.dir/all] Error 2
CMakeFiles/Makefile2:156: recipe for target 'firmware/CMakeFiles/firmware.dir/rule' failed
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
Makefile:131: recipe for target 'firmware' failed
make: *** [firmware] Error 2

ImDroided, 2019-11-07:
Maybe update your hackrf installation. I'm at work. When I get home I'll do
a pull and see if I have the same issue

infodel, 2019-11-07:
Ok... Starting with a new Pull request.... I get this while running cmake.. 

cmake ..
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr/arm-none-eabi
CMake Error at CMakeLists.txt:48 (add_subdirectory):
  **add_subdirectory given source "hackrf/firmware/hackrf_usb" which is not an
  existing directory.**

-- Configuring incomplete, errors occurred! 
See also "/home/osboxes/portapack-havoc/v2/portapack-havoc/build/CMakeFiles/CMakeOutput.log".
See also "/home/osboxes/portapack-havoc/v2/portapack-havoc/build/CMakeFiles/CMakeError.log".

I then took the compiled hackrf_usb directory that i got from the hackrf github and placed its dir where it needed and cmake finished without any issues... I then got the error previously during the build process at the other user.

ImDroided, 2019-11-07:
Rename your old directories and do a fresh git clone on both havoc and
hackrf. I remember last time I had to do this and it fixed all my errors I
was having.

furrtek, 2019-11-08:
See #274.

closed 2019-11-08 by furrtek