furrtek/portapack-havoc #274: problem compiling in ubuntu 19.10

sgoudelis, 2019-11-07:
I did do:

```
sgoudelis@mybox:~/projects/portapack-havoc$ git submodule init
Submodule 'hackrf' (https://github.com/jboone/hackrf.git) registered for path 'hackrf'
sgoudelis@mybox:~/projects/portapack-havoc$ git submodule update
Cloning into '/home/sgoudelis/projects/portapack-havoc/hackrf'...
Submodule path 'hackrf': checked out '923f9fe617c9dcf00c97c18f5aa79e51720b9e59'
```

And I am getting this:

```
[ 18%] Linking CXX executable baseband_sstvtx.elf
[ 18%] Built target baseband_sstvtx.elf
Scanning dependencies of target libopencm3_hackrf_usb
[ 18%] Creating directories for 'libopencm3_hackrf_usb'
[ 18%] No download step for 'libopencm3_hackrf_usb'
[ 18%] No patch step for 'libopencm3_hackrf_usb'
[ 18%] No update step for 'libopencm3_hackrf_usb'
[ 20%] No configure step for 'libopencm3_hackrf_usb'
[ 20%] Performing build step for 'libopencm3_hackrf_usb'
make[4]: *** No targets specified and no makefile found.  Stop.
make[3]: *** [hackrf/firmware/hackrf_usb/CMakeFiles/libopencm3_hackrf_usb.dir/build.make:111: hackrf/firmware/hackrf_usb/libopencm3_hackrf_usb-prefix/src/libopencm3_hackrf_usb-stamp/libopencm3_hackrf_usb-build] Error 2
make[2]: *** [CMakeFiles/Makefile2:328: hackrf/firmware/hackrf_usb/CMakeFiles/libopencm3_hackrf_usb.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:699: firmware/CMakeFiles/firmware.dir/rule] Error 2
make: *** [Makefile:313: firmware] Error 2
```

Any ideas?

ImDroided, 2019-11-07:
Looks like it's missing a library did you follow the procedures to install
the hack RF for your computer?

sgoudelis, 2019-11-07:
Now that you mention that I havent updated the hackrf stuff on my box. I though this repo did not require anything else.

Do you mean the hackrf library or something else ? It seems to me that getting this library was part of the make procedure. Isn't that so ?

ImDroided, 2019-11-07:
It's been so long since I installed I'm not exactly sure I'm just reading
the error message basically saying that it can't find a library maybe try
to refresh the hack RF stuff see if that fixes it otherwise Google that
library and see how to install it but I'm pretty sure when I installed my
hack RF it took care of everything.

z4ziggy, 2019-11-08:
git submodule update --init --recursive

sgoudelis, 2019-11-08:
Yeah, this seems to work. Maybe put this up in the docs ?

closed 2019-11-10 by sgoudelis