furrtek/portapack-havoc #282: (Capture & /FREQMAN related) Saving files seems to be an issue

Ahorrorfilm, 2019-12-01:
I can save a frequency within [Capture] and give it a name, however when that frequency is saved it overwrites the files within the first folder "ACARS" within the /FREQMAN folder.

lillstrumpa, 2020-01-13:
Had the same problem, a quick fix is to make a new folder with the name "1" then everything new without name is saved in that folder, or replaces the file in their...

Ahorrorfilm, 2020-01-13:
That's actually what I did haha, I wonder how you could write it in to tell it to save captures in a "Capture" specific folder. I know this workaround has been working but I'm all about a clean interactive UI. Forked it so I can see what's going on or if anything needs revised, if that's even anything I'm capable of. Furrtek has his hands full and it's a wonder hes even made a masterpiece like this haha. I'll keep ya posted if I find anything if he doesn't get to it before me...he probably will. ;D

Ahorrorfilm, 2020-01-19:
Okay so it looks like the issue resides in changed files. When you rename a recently captured file you'll also notice it doesn't retain its .C16 format. Still experimenting with it albeit very slowly as I've been working heavy hours throughout the week, hopefully I can free up some more time but till then I thought it helpful to provide some feedback along the way. 

┬┴┬┴┤ ͜ʖ ͡°) ├┬┴┬┴