furrtek/portapack-havoc #283: Make it completely usable without touchscreen support

bryanmcguire, 2019-12-09:
There is a flood of Chinese units that don't have a working touchscreen. This means that you can't delete a directory on an SD card, for example. One fix could be to making it so pushing the right button goes to the menu instead descending to the selected directory. Currently the right button and center button perform the same action, so this change wouldn't cause a loss of functionality (though I admit it could be considered a degradation of user experience for those with functioning touchscreens).  Maybe it could be an alternate setting?

furrtek, 2019-12-17:
Could be an easy fix, but that would still be effort to work around a problem that shouldn't exist.

They didn't spend time making the touch screen work and they don't even respect the license, why would anyone have to spend even 5 minutes doing their work ?

Just return the faulty Chinese Portapacks, they're the ones making money, they're responsible for what they sell.

gregoryfenton, 2020-04-11:
For those (such as myself) that have purchased the cheap Chinese unit the issue is a cantfix rather than a wontfix from the developers of the software - there is physically no touchscreen mounted to the board.
There are sellers online that sell the touchscreen part on its own or as a complete unit including LCD display. This is the option I have taken. You need to search for either:
er-tp024-3 for the touchpad alone; or
 er-tft024-3 for the screen. Make sure you get the screen with the touchpad already mounted, it will be simply a case of unclipping the old ribbon cable, removing the screen (it is attached with double sided tape) and then mounting the new screen in the same way and position.