furrtek/portapack-havoc #284: [Quiestion]Connect and software-enable speaker

Radiomix2000, 2019-12-15:
There is place for speaker plug on PCB, which is connected to AK4951. Is it possible to use portapack with speaker and how to activate it in firmware?

DinBan, 2020-01-10:
Any one can answer to this good question? Thank you!

cql1983, 2020-03-21:
I looked at the source code,，
`firmware\application\audio.cpp `

`
namespace output {

void start() {
	i2s::i2s0::tx_start();
	unmute();
}

void stop() {
	mute();
	i2s::i2s0::tx_stop();
}

void mute() {
	i2s::i2s0::tx_mute();
	audio_codec->headphone_disable();
}

void unmute() {
	i2s::i2s0::tx_unmute();
	audio_codec->headphone_enable();
}`

At
`firmware\common\ak4951.cpp`
the speaker_enable() function Implemented this program
It includes the initialization process of opening headphones. I think it can be simply edited to initialize the speakers at the same time. But if you want to add the open and close menu in the program, it needs some work.

RChadwick7, 2020-03-23:
Using a H2 with an amplifier IC installed, I was able to get audio without issue.

cql1983, 2020-03-25:
> 
> 
> Using a H2 with an amplifier IC installed, I was able to get audio without issue.
The ak4951 originally supports external speaker output, and the official did not launch H2. H2 is the amplification chip added by some fans' DIY, not the officially supported hardware.

jamesshao8, 2020-05-02:
Hi 
I have added a button on top to switch on/off the speaker, you can try it in my repo.