furrtek/portapack-havoc #285: cannot create files outside of havoc firmware and replay them with portapack

jfender1, 2019-12-15:
I am unable to create files in gnu radio and have them play in the portapack. I would like to be able to use filters to quite down the signal to get a very clean and precise signal and have the portapack replay them. It seems that if I just change the file extension to c16 it will show up but it doesnt seem to play correctly or the graphics look very weird. How do I look at the c16 captures? I would like to see how clean the signal is being recorded by the portapack.

furrtek, 2019-12-18:
Have you seen this ? https://github.com/furrtek/portapack-havoc/wiki/Capture

v1tog, 2019-12-20:
Hi i don't know if this is the right place to put this question but i'm one day new to this site and one week with my hackrf one so here is my question can someone please point me in the right Direction on how to put files on a SD card for the portapack ie GPS simulator and world maps thank you in advance

rightrozz, 2020-01-10:
Using csdr or audacity to convert to complex signed 16 bit audio won't correctly play on the hackrf for me.  Has anyone got a successful method on converting files to C16 format?

jamesshao8, 2020-02-14:
I have similar problem as you, I modified replay in portapack to send a data file (2MHz 8 bit), which I can send using gnuradio using ichar to complex. However there is still problem.

cosmo6110, 2020-03-07:
I using audacity to edit c16 files
import raw : signed 16 bit pcm, little-endian, 1 channel, sample 44100
to save i use ffmpeg function.
export audio,  open ffmpeg, format s16le codec pcm_s16le export ffmpeg save