furrtek/portapack-havoc #291: Putting files on SD card for portapack

v1tog, 2019-12-20:
Hi i don't know if this is the right place to put this question but i'm one day new to this site and one week with my hackrf one so here is my question can someone please point me in the right Direction on how to put files on a SD card for the portapack ie GPS simulator and world maps thank you in advance

furrtek, 2019-12-20:
Hello. Asking the question once is enough.
See https://github.com/furrtek/portapack-havoc/tree/master/sdcard
The "GPS simulator" is just a hacked up replay screen only found on Chinese Portapack clones, ask the seller.

v1tog, 2019-12-20:
Hi i'm new to this hackrf one but is there any videos showing how to put files on a SD card and making them run on the portapack and the link you provided I have downloaded but i don't no how to get it to work on the SD card thank you

furrtek, 2019-12-20:
It's written in plain text, no need for a video. Just copy the files to the root of the SD card.
For the world map, you need to generate it as explained.

v1tog, 2019-12-20:
OK thank you for your time i will suss it out one day cheers

closed 2019-12-21 by v1tog