furrtek/portapack-havoc #299: Pocsag stopped working after reinstalled frimware

pawel37, 2019-12-31:
after reinstalling firmware pocsag stopped work on receiving or transmittining.  There is 3 ver of the firmware out one with the drop dead working one with icons not text on the main menu and ur one .
The icon one can Rx l  not Tx . Drop dead can Rx but not Tx . Ur one cant Rx or Tx now before it could like 3 weeks ago 

Thank you 
Paul

g-c-n, 2019-12-31:
Dude.

furrtek, 2019-12-31:
Then put back the version you had 3 weeks ago.

pawel37, 2020-01-01:
it was ur version 3 weeks ago now not working

RChadwick7, 2020-01-01:
I'm probably going to regret this...
Do you know exactly what version worked? How is this version not working? Did you compile both versions? Any other details you can offer? Just saying it doesn't work isn't helpful. If you had a problem with your car, and brought it to a mechanic, the first thing they will ask is what the problem is. Just saying it doesn't work isn't helpful.

If all else fails, you can just ask for a refund for the price you paid furrtek for the firmware.

pawel37, 2020-01-01:
about 3 or 2 weeks i download it from furrtek  github and pocsag rx and tx was working now his version not rx ot tx on pocsag .i can hear it on a scanner but nothing can up on the portapack . i change to the drop dead one i can see massage on the portapack. but not tx to a pager i can hear it on the scanner.
i did not compile any of them downloaded the bin file

RChadwick7, 2020-01-01:
Do you have the link to the bin file(s)?

pawel37, 2020-01-01:
https://www.mediafire.com/file/kpi4a550iux9fhk/portapack-h1-havoc-vma.bin/file

https://github.com/furrtek/portapack-havoc/tree/master/firmware

furrtek, 2020-01-01:
> If all else fails, you can just ask for a refund for the price you paid furrtek for the firmware.
Pffrt thanks my day got a bit better.

@pawel37: there's a singe .bin file in this repo. I don't know where you're seeing 3.
The pocsag code for rx and tx hasn't changed in more than 8 months, as you can see for yourself by searching for "pocsag" in this repo and checked the last commit date.
If you want me to investigate, at least get your infos straight. Flash the only bin I provide and tell me what's going on and what settings you're using.

pawel37, 2020-01-01:
ok i will but i cant flash now flashed a bin file and not stopped working 
https://github.com/furrtek/portapack-havoc/issues/78#issuecomment-318949523 thats the bin file i flashed

[image]

furrtek, 2020-01-01:
"And not stopped working" ? What... ?
Hardware problem, no 1.8V and USB LED doesn't look right.
That bin isn't from me, you're on your own with it.

RChadwick7, 2020-01-01:
So the first firmware you had was the Mediafire link, then 2 or 3 weeks ago you got the latest bin from github?

RChadwick7, 2020-01-01:
Well, it looks like problem solved. Sorry about your hardware issue.

closed 2020-04-06 by furrtek