furrtek/portapack-havoc #301: Question - Battery indicator?

DinBan, 2020-01-10:
Hello, is it possible add to firmware battery menu with charge level?
I think it is useful thing for portapack, it is designed for use it with battery as mobile device.
Thank you in advance!

furrtek, 2020-01-10:
Hello. The Portapack can't be aware of the battery level as it only receives steady 5V through the USB port. It can't even know if it's powered by a computer or a charger, or a battery.
A typical USB power bank will provide a steady 5V as long as it can supply enough current, and then shut right off when it can't anymore, without giving warnings through the USB port.

RChadwick7, 2020-01-10:
This is related to another question I was going to ask. Is Havoc compatible with the Portapack H2? It appears to have it's own internal battery.
Also, a few of us have managed to squeeze an internal battery into our Portapacks, but I'm sure there's no standard way of detecting voltage, unless if one of the Microcontroller pins has a nice A/D converter. Hey, if you add it in the software, I'll add it in my hardware :)

DinBan, 2020-01-11:
I bought from Aliexpress portapack h1 with battery, it labeled 1500mah, but I think it is not. Thats why I'm asking. It charges good when it connected to USB or charger, but I never know when it charger for 100%. OK, thank you for replay. Will do it in hardware way... Best regards!

furrtek, 2020-01-11:
Well... that's Aliexpress unfortunately.

eried, 2020-03-30:
> It charges good when it connected to USB or charger, but I never know when it charger for 100%

It is the typical charger controller clone they put on all power banks. The leds goes off but I think in the H2 it blinks in a glitchy way (probably because the portapack is consuming energy?) but in conclusion: when it is 100% the charging led blinks erratically.

eried, 2020-04-06:
UPDATE: The led does not blink erratically if the CR1220 is in place on the H2.