furrtek/portapack-havoc #303: QUESTION - SWR meter option possible?

DinBan, 2020-01-19:
Hello, is there any way to add to HAVOC firmware SWR/VNA menu/option in future?
I found that HackRF/SDR can be user as SWR meter with cheap noise source -
https://www.rtl-sdr.com/measuring-the-swr-of-fpv-antennas-with-an-rtl-sdr/
Also found on GitHub some hackrf_antenna_swr_utility - 
https://github.com/coddingtonbear/hackrf-antenna-swr-utility/tree/master/hackrf_antenna_swr_utility
I think it would be nice and useful option if it possible.
Thank you!

unixpunk, 2020-01-19:
IMO, this would be a good request for the main portapack (non-havoc) repo - https://github.com/sharebrained/portapack-hackrf

DinBan, 2020-01-21:
Thanks for your opinion

> IMO, this would be a good request for the main portapack (non-havoc) repo - https://github.com/sharebrained/portapack-hackrf

Thank you for your opinion. Just ask same question at main portapack (non-havoc) repo.
https://github.com/sharebrained/portapack-hackrf/issues/148

Hope someone will replay to this question here or at non-havoc repo.

Best regards!