furrtek/portapack-havoc #305: Feature request: GPS Simulator

tommy5467, 2020-01-30:
Would it possible to add a GPS Simulator function? I saw some Chinese clones have it in some firmware. THX.

furrtek, 2020-01-30:
The Chinese clones "GPS Simulator" menu item is just a barely modified version of the IQ file replay app.

tommy5467, 2020-01-30:
Thank you so much for your time!

closed 2020-01-30 by tommy5467

jamesshao8, 2020-02-14:
Hi, I am trying to duplicate this (gridrf version). I modified replay myself, since they are not open source.I made replay working with C8 and sample rate of 2MHz. However, when I was trying to send out the signal, the spectrum is a little different and cell phone can't lock onto that signal.

Can you give me some advice? I will post my code on github. I hereby post the spectrum with gridrf transmitting and my own (4x 8x oversample rate).
GridRF
[image]
Mine 4x
[image]
Mine 8x
[image]

When I was using 4x sample rate, my cell phone can receive some of the gps signal, but can't lock. When using 8x sample rate, none of the satellite signal can be received.
[image]

jamesshao8, 2020-02-14:
https://github.com/jamesshao8/portapack-havoc-modified/blob/master/firmware/baseband/proc_gps_sim.cpp
https://github.com/jamesshao8/portapack-havoc-modified/blob/master/firmware/baseband/proc_gps_sim.hpp
https://github.com/jamesshao8/portapack-havoc-modified/blob/master/firmware/application/apps/gps_sim_app.cpp
https://github.com/jamesshao8/portapack-havoc-modified/blob/master/firmware/application/apps/gps_sim_app.hpp

This is my modification of the code. Please give me some advice. Thank you.
@furrtek

jamesshao8, 2020-02-14:
nevermind, I have already fixed it.

BreakSecurity, 2020-02-14:
@jamesshao8 please try to do a pull request ;)

jamesshao8, 2020-02-15:
Hi @BreakSecurity I dont know how to pull my code here, I guess it requires furrtek's consent, right?
You can check the code in my repository:
https://github.com/jamesshao8/portapack-havoc-modified
or you can directly try my firmware here.
portapack-h1-havoc-gps.zip