furrtek/portapack-havoc #306: Capture bug

jamesshao8, 2020-02-14:
When portapack is powered, I can't enter capture section directly. The device is blocked and can't react to any button/touch.

I need to firstly use some receiver like am receiver, then exit. After that, capture can be entered normally.

This happens both on the latest version and the old version.

By old version, I mean the one with this main screen.
[image]

furrtek, 2020-02-14:
Some icons are missing. What version is that ?

jamesshao8, 2020-02-14:
@furrtek  Thanks for your reply. That screenshot is from rtlsdr's website. My portpack menu looks similar to that, so I didn't take the screenshot by myself back then.
Now I took a look at the version(git hash), it's 30db228. This is the GridRF version (Chinese version with the gps simulator). I suspect it was because they modify the code in a wrong way? However, the latest code on the master branch in your repository has the same problem. Its git hash is UNKNOWN.

The main menu of two versions I have tried:
[image]
[image]

jamesshao8, 2020-02-14:
I also tried the firmware you provided in firmware directory, git hash (dc5d6fe). There is no this kind of problem on that version. So this shouldn't be a problem of hardware.

jamesshao8, 2020-02-14:
I hereby uploaded the firmwares that have this problem.
One of them is the latest code from master branch, the other is the GridRF(Chinese GPS) version.
[bug.zip](https://github.com/furrtek/portapack-havoc/files/4204490/bug.zip)

x2plag, 2020-02-14:
> I hereby uploaded the firmwares that have this problem.
> One of them is the latest code from master branch, the other is the GridRF(Chinese GPS) version.
> [bug.zip](https://github.com/furrtek/portapack-havoc/files/4204490/bug.zip)

Hello. Can you please describe that GridRF firmware? Is the only difference is that they added GPS emulation? Does it works for china only?

jamesshao8, 2020-02-14:
@x2plag Hi, GridRF firmware is a modified version of older havoc firmware, I think it's around June, 2019. They added a section similar as replay to send out gps iq files produced by gps-sim-sdr. It should work globaly. However, they didn't make it open source. So I recently written a gps mode in havoc firmware based on the same idea. And put it open source.
You can check the code in my repository:
https://github.com/jamesshao8/portapack-havoc-modified
or you can directly try my firmware here. 
[portapack-h1-havoc-gps.zip](https://github.com/furrtek/portapack-havoc/files/4205048/portapack-h1-havoc-gps.zip)

Flying-Pit, 2020-02-15:
@jamesshao8 , i wanted to test your firmware, but my portapack crashes. same with the other two firmwares. After DFU recovery and loading the Havoc firmware it works again. What can be wrong? I just wanted to test your firmware.
Greetings

jamesshao8, 2020-02-16:
Hi @Flying-Pit By the other two firmwares, you mean those two I put in bug.zip? One of them is just the latest build of furrtek's code and shouldn't make portapack crash. You can download the code and build by yourself. When I am flashing firmware, I never have to use dfu recovery. Simply switching to hackrf mode on the menu, and using hackrf_spiflash is enough. Can you tell me which version of havoc firmware is working on your side? What's your procedure of flashing the firmware?

Flying-Pit, 2020-02-16:
hi @jamesshao8, yes this two i mean. they not works. also ur´s GPS FW want not work, i get an Error if i flash this all FW´s. Then i must flash back the original hackrf_one_usb.bin and after i flash the last HAVOC FW i found. ( GitHash:dc5d6fe) this its the only one wich works fine. 
I use a chinese Portapack  but i think its not the Hardware. And..i use Windows, not Linux, and i have no experience to make a Code with Windows. I only use ready code (bin´s) and i use for flashing the 
"portapack.firmware.tool20190405" this works fine under WIN7. Im a little bit confused.
Thanks for helping me!

Flying-Pit, 2020-02-16:
so, is also flash quickly the latest FW from "sharedbrain" 20190405, works fine. and then i flashing the GPS FW from you, ERROR, the Portapack/HackRF hangs and i must start the first Procedere again. 
i hope this helps.

jamesshao8, 2020-02-21:
Hi, @Flying-Pit I have tried git hash dc5d6fe, it doesn't have the capture bug indeed. This is a old version of havoc firmware. I modified the firmware based on the newest firmware, which itself have this bug. In the case of your hang up problem. I really don't know why, as I rarely use windows.

closed 2020-06-24 by jamesshao8