furrtek/portapack-havoc #308: the screen portapack not working

Mars0117, 2020-02-26:
Hi!
Earlier i installed the portapack-havoc firmware. 4 day worked. And then I tried SSTV for the first time, pressed "start" and the screen not working.
And i installed the portapack-sharebrained, also not working...
I re-installed the portapack-havoc firmware. I tried with the DFU bootloader but the of no use:

dfu-util --device 1fc9:000c --download hackrf_one_usb.dfu --reset
hackrf_spiflash -w portapack-h1-havoc.bin

also not working...

[image]
After connecting USB to a PC or PowerBank the 3V3 and the 1V8 and the RF is lights always ON, USB and TX is always OFF, RX yellow light always blink.

hackrf_info:
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000057b068dc25488963
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00594766

Thank for your help

furrtek, 2020-02-26:
Hello,
Chinese clone, doesn't work with the original firmware from Sharebrained -> fault unrelated to firmware. Closing.

closed 2020-02-26 by furrtek

Mars0117, 2020-02-26:
Before the problem the original firmware and havoc worked 4 day.