furrtek/portapack-havoc #312: CLKOUT signal not present?

makkiato83, 2020-03-08:
Hello,

I have noticed that when in portapack mode, my hackrf does not emit a 10Mhz signal (CLKOUT).
The signal starts only when I quite Havoc and go in "HackRF mode".

Is it possible to activate it from somewhere inside Havoc? 
If not, would it be possible to provide this option or there is some technical obstacle?

Similarly (but I haven't been able to check this), does the Portapack automatically uses an External 10MhZ signal?

Thanks

eried, 2020-04-27:
Mine emits the clock in portapack mode. But I have one of those H2 portapacks with a soldered external 0.5 %ppm clocks

fcanaan, 2020-05-06:
I have a clone H1 Portapack and can verify mine does not emit the clock when in portapack mode either. I know the portapack that I have has a 10mhz TCXO .5ppm soldered onto it. When i switch to hackrf mode, the clockout begins working.

eried, 2020-05-07:
In portapack mode:
[image]

In hackrf mode:
[image]

I can see in hackrf mode the signal is stronger tho.