furrtek/portapack-havoc #317: create a custom splash image

jamesshao8, 2020-03-28:
I am experimenting to change the start logo to a custom image.

I didnt littlle bit research, it's in /firmware/application/bitmaps/bmp_splash.hpp
Also I found bmp_modal_warning.hpp near that file. And I find the modal_warning.bmp in firmware/graphics

Now when I didnt the following command, I can get the same result as in  bmp_modal_warning.hpp
" xxd -i modal_warning.bmp modal_warning.h"

And if I replace the value in bmp_splash.hpp, the start logo can be changed.

However, if I use a custom bmp file, I can't generate valid data.

What specific type of bmp is suitable to be used here?

Ahorrorfilm, 2020-03-28:
Check the size of the last bmp image and then convert yours to that size, lemme kno if that doesnt work out.

furrtek, 2020-03-29:
Hi @jamesshao8.
You need a 4-bit RLE compressed bitmap, I recall it was possible to save such a file with GIMP, look for "RLE4".
For more details, check out `ILI9341::drawBMP` in `lcd_ili9341.cpp`.

jamesshao8, 2020-03-29:
Thanks, I will give it a try.

closed 2020-04-30 by jamesshao8