furrtek/portapack-havoc #319: GPS Module

catalinalb, 2020-03-29:
By adding the GPS module will correct the clock automatically? Can someone share more info on this topic?

Thank you,
Catalin

eried, 2020-03-30:
Can you share which GPS module are you adding :P

DinBan, 2020-04-03:
Very interesting...

Dirk275, 2020-04-06:
Catalin possibly refers to the GPS module, that could be soldered on the PortaPack H1.
The uBlox MAX-M8 fits to this place.
But adding this module would mean a firmware support, which actually doesn't exist.
Anyone interested in adding this module and/or firmware support?

eried, 2020-04-06:
I would be interested but I am just getting familiar with the firmware right now

Dirk275, 2020-04-07:
eried: me too, I just have my PortaPack for a few weeks.

Catalin: the advantages of a built-in GPS module could be:
-- automatic setting/correcting of time and date
-- knowing the own position (for example to center the map to the own position in ADSB and not only to the plane, ...)
-- using frequency tables fitting to the area you are in
-- automatic ppm correction for the HackRF One receiver (best advantage, I think!)
-- ...
But: each of these "features" needs a firmware support (must be written ...).

catalinalb, 2020-04-19:
Hello,

Yes, I do refer to the GPS module that can be soldered on the board. And yes I would like to buy the MAX-M8 but there are 3 versions available: 
https://www.u-blox.com/en/product/max-m8-series

 MAX-M8C, MAX-M8Q, MAX-M8W

Are there any firmware developed for the GPS module? Would be a great improvement to have the time accuracy from the GPS since the internal clock of hackrf is usually 20ppm accuracy.

If that´s working on the CLKOUT of the Hackrf we should a perfect 10 MHz clock signal so the unit can be used as a GPSDO too.

If anyone has experience/knowledge, please let us know.

Thank you,
Catalin

Dirk275, 2020-04-19:
Hello,

no experience, but looking at the Portapack schematic I see the TIMEPULSE output pin of the GPS module is not left empty, but connected with the main IF part.
So: maybe GPSDO is possible ... 
Dirk