furrtek/portapack-havoc #326: Another World Map for ADS-B

Dirk275, 2020-04-08:
In northern Europe the existing world map consists mainly of clouds or snow, so that almost nothing can be seen on the map in ADS-B mode.
For a better use would it be possible to use another world map? I would prefer a map with at least country boundaries, the bigger airports and perhaps some rivers or main roads for orientation.
For me there would be no need for an original partial cloudy satellit image for this purpose. The world map project II has some good artificial world maps.

If another world map could be used: are there certain things to take care of for another jpg map? E.g. the map projection type, same pixel dimensions, 24 bit graphics, color table, dpi or else ...

eried, 2020-04-27:
There is instructions to generate the map https://github.com/furrtek/portapack-havoc/tree/master/sdcard have you tried to generate another map?

Dirk275, 2020-04-28:
Yes, I tried.
But didn't work. Perhaps I didn't match the original jpg data:
21600x10800 pixel, 96 dpi, 24 bit color depth.

eried, 2020-04-28:
What didn't work? Which part failed? I could take a look on the code that reads the image to help trying to get your map but of course the first step would be to match it perfectly, then change 1 thing, retry etc

Dirk275, 2020-04-28:
Hi eried,
thank you very much for your answer!

I used THIS artificial kind of geographic world map as a starting point and as a 1st test of another map:
[https://www.dropbox.com/s/czzbvlhujlwnscn/world_map.jpg?dl=0](url)
... because it has the same size (21600x10800) and seems to have the same projection type. The map is not what I would like to have, but perhaps it could be a starting point for adding country boundaries, ...
But maybe other parameters of that picture don't match ...
It seemed to compile, but didn't show on Portapack screen in ADS-B.

eried, 2020-04-28:
It might be the resolution?

[image]

Because I can see that metadata has slight differences in 3 bytes but I do not think that will produce any problem
[image]

Dirk275, 2020-04-28:
Ok, nearly no difference between those pics.

If you compile the other jpg: is it displayable in your Portapack?
I'm on FW 79baef7, which is from 12-29-2019, I think.

eried, 2020-04-28:
Resolution will make a big difference tho, I will do some tests later

eried, 2020-04-28:
> Ok, nearly no difference between those pics.
> 
> If you compile the other jpg: is it displayable in your Portapack?
> I'm on FW [79baef7](https://github.com/furrtek/portapack-havoc/commit/79baef7731e5b66257a513e7a03859e2bbdfb0f6), which is from 12-29-2019, I think.

The bin file is quite smaller with your map. I also live on north europe, so I hope I can use that one too lol, I am still waiting for some activity to see how the map works on my portapack. If it works I will upload that bin

eried, 2020-05-02:
OK I figured out the problem, the world_map.py script that generates the bin is wrong. You should have something like this:

               # RRRRRGGGGGGBBBBB
                pixel_lcd = (pix[x, y][0] >> 3) << 11
                pixel_lcd |= (pix[x, y][1] >> 2) << 5
                pixel_lcd |= (pix[x, y][2] >> 3)
                #         RRRGGGBB to
                # RRR00GGG000BB000
                #pixel_lcd = (pix[x, y][0] >> 5) << 5
                #pixel_lcd |= (pix[x, y][1] >> 5) << 2
                #pixel_lcd |= (pix[x, y][2] >> 6)
                line += struct.pack('H', pixel_lcd)

I have not tried yet but this is what I think after fiddling around few hours with the map (I will make a Mercator map projection using openseamap+openstreetmap). 

I will push the corrected .py if that was the solution.

Dirk275, 2020-05-04:
Ok, good that the problem can be solved!
The actual map is an equirectangular projection.
If you change the projection to Mercator, you will have to change the ADS-B "app" code too, because the calculation of GPS positions in the map refers to the equirectangular map.