furrtek/portapack-havoc #327: PortaPack H2

lagorcejb, 2020-04-10:
Hi,

i have receive today my new portapack H2. Realy a good upgrade !!!
I have build my own firmware for my old H1. but what about the H2 ?
the firmware indicate number : 30db228

Can i juste build like a normal H1 ?
do i have to change somthing about the screen resolution ?

thanks !

furrtek, 2020-04-10:
Hi. How about trying it ?
Also H2 doesn't respect the H1's license, so no support (at least from me).

lagorcejb, 2020-04-10:
It will be pleasure but i dont know where to find the original firmware.
So if i brink the device i cant go back :) ...

Thanks by the way.

furrtek, 2020-04-10:
You can't brick it, the DFU is in ROM.

lagorcejb, 2020-04-10:
Yes but If i dont have the original firmware i cant rollback my device no ?

lagorcejb, 2020-04-11:
Well it's working ... I have put the default firmware the H2 is ok.

I need to test with bluescreen when you use as USB.
Cause for now when i go to USB i have a balckScreen.

closed 2020-04-11 by furrtek