furrtek/portapack-havoc #331: Headphone Failure

SannhetJaeger, 2020-04-15:
Think I am dealing with a defective portapack, but thought I would see if anyone has any ideas on steps to take to remedy or be sure its a hardware failure.  Headphones produce no output, regardless of volume level.  I've tried with several different firmwares (20170808, Havoc 1352e16 and a locally built Havoc from few days ago).  Right now have the 20190405 installed.  This is a 20181029 Blue Portapack board on a 10 Feb 2014 HackRF One.  Debug of AK4951 shows all zeros(0).

Only reaction I can get is on microphone transmit - pressing the apple headset button makes the mic tx level bar come on if it wasn't there already (e.g. check / uncheck voice activated will make bar disappear).  Unfortunately, that's only response - no actual mic sounds are picked up.  Thought it relevant to share that oddity as it shows "something" is getting through.

The headsets are confirmed working on other devices.  Used several other brands of TRRS (left, right, ground, mic) headsets.  Also tried TRS (left, right, ground) to no effect.  Confirmed continuity from each jack to jack mount pads to next circuit step (e.g. left-C22, right-C23, mic-C25, ground).

Thanks in advance!

eried, 2020-04-18:
Do you have another portapack to test? The clones are quite cheap in aliexpress and a backup wont harm.

BTW, I dont think this issue is really related to havoc

SannhetJaeger, 2020-04-19:
@eried Yep, think that's what I am going to do.  Seller offered to exchange, but that will be weeks of down time.  Now its just become a puzzle I must solve, so digging deep into circuit a rewarding experience.

N0vaPixel, 2020-04-20:
@SannhetJaeger I suspect your portapack to be a chines clone. Can you confirm that yours looks exactly like the one on the sharebrained website ? (especially the silkscreen). [https://sharebrained.com/](https://sharebrained.com/)

Furthermore, furrtek/portapack-havoc is a repository related to firmware development. So don't expect many interesting replies related to an hardware issue. See peppa pig's advises in [README.md](https://github.com/furrtek/portapack-havoc/blob/master/README.md)

Best bet is to ask to the seller. To reassure, you are pretty safe if you bought your portapack from sharebrained or a re-seller.

furrtek, 2020-04-20:
Yup, seller selling crap -> seller's problem.
The firmware does have a ton of bugs but something as basic as audio output and touchscreen support **must work**.

SannhetJaeger, 2020-04-21:
Indeed, it's produced in China.  For a hobby item, it fit the budget.  Do appreciate the feedback though.  Engineering wise, it matches exactly to KiCad schematics / footprint.  Under microscope, it's quite clean.  Didn't have enough jumpers to flip board over to test with logic analyzer - waiting of snail mail to deliver more.  Since all other functions work and the radio is well calibrated per scope, this will just be a pet project to learn on.  If it can be built, it can be fixed :-).

In the meanwhile, I am figuring out how to get some insights into the init sequence.  I'll look through the code to see about re-enabling the debug screen (saw another person asking about this) as an option from settings, with perhaps some test functions with captured register states; maybe these test routines are in older versions of firmware.  Not sure if there are ways to do this interactively from CLI when in hackrf mode?

closed 2020-04-21 by SannhetJaeger

SannhetJaeger, 2020-04-24:
FYI - turns out idiots that manufactured board put the AK4951 chip on board 90 degrees rotated.  Think that might cause issues?  Rotated chip and its working as expected...  Lesson #90 +/- 2M, check for chip pin 1 placement early in troubleshooting - way before reflowing and logic analyzing board...

furrtek, 2020-04-25:
Wow that's some great QA...