furrtek/portapack-havoc #347: Enhancement request: Allow USB access to SD card in HackRF mode

gregoryfenton, 2020-04-25:
I would love to be able to edit the SD card while it is plugged into the PortaPack.
I've been bitten in the past with micro SD cards flying out of the slot and disappearing forever when attempting to remove them.
Is there any chance of allowing read and write access to the SD card in HackRF mode?

furrtek, 2020-04-25:
Would require implementing the Mass Storage class and provide a way to switch between IQ stream and drive mode. Having both at the same time (if even possible ?) would be a mess.

eried, 2020-04-27:
> I've been bitten in the past with micro SD cards flying out of the slot and disappearing forever when attempting to remove them.

My temporal solution was to add a piece of tape, but I agree this would be nice feature.