furrtek/portapack-havoc #348: Devices detected: 0 DFU, 0 HackRF ERROR: HackRF not found. Please connect a HackRF One to your computer's USB port. If using a PortaPack, please put it in HackRF mode after connecting it.

andytang123456789, 2020-04-30:
I received the radio through #SDR, but I cannot flash the firmware because the firmware cannot recognize hackrf one, even if I enter hackrf mode

Phreak87, 2020-04-30:
Have you done the Installation via zadig/urules.dev?
I know this Problem If the Driver was wrong or vbox cannot Import the device.

andytang123456789, 2020-04-30:
yes  via zadig ,i I received the FM radio through #SDR  on the computer

Phreak87, 2020-04-30:
Are the drivers installed correctly? 
Zadig winusb or Linux udev.rules?
In vbox you have to auto-mount the dfu-device via the settings

Phreak87, 2020-04-30:
The dfu is another device. Please Check via the device Manager / zadig again If the dfu Driver is installed

andytang123456789, 2020-04-30:
I can enter DFU mode and the computer recognizes it

Devices detected: 1 DFU, 0 HackRF
WARNING: One or more DFU devices detected.
Updating from DFU mode is not yet supported.
ERROR: HackRF not found.
Please connect a HackRF One to your computer's USB port.
If using a PortaPack, please put it in HackRF mode after connecting it.

eried, 2020-04-30:
Go to device manager and replace the driver with the one in /driver/ in the firmware zip

andytang123456789, 2020-04-30:
sorry , do you have it firmare and driver  ,would send it to my email :fcstrend@gmail.com

andytang123456789, 2020-04-30:
[image]

eried, 2020-04-30:
Go to device manager and replace the driver with the one in that zip

[image]

If the device is not there, check http://blog.didierstevens.com/2017/09/18/quickpost-gnu-radio-on-windows/

andytang123456789, 2020-04-30:
i did it but ,it is still  have the problem , maybe the some other problem ,
[image]
[image]

andytang123456789, 2020-04-30:
and did it according this vedio not successful also
https://www.youtube.com/watch?v=5G5blPPIIBo

eried, 2020-04-30:
Did you installed the driver updating "HackRF One"?

eried, 2020-04-30:
[image] 

Select `driver/hackrf_one.inf`

andytang123456789, 2020-04-30:
YES ,I did it ,and My computer says this is the best driver

eried, 2020-04-30:
> YES ,I did it ,and My computer says this is the best driver

But select `driver/hackrf_one.inf`, not automatic.

andytang123456789, 2020-04-30:
it is right  ？[image]

andytang123456789, 2020-04-30:
sir, problem solved , have to use  old version  
[image]

andytang123456789, 2020-04-30:
thank  you sir , for you support

closed 2020-05-24 by furrtek