furrtek/portapack-havoc #350: Audio App crashes Portapack

helmarw, 2020-04-30:
when switching from Audio app to e.g. TPMS and then back to Audio, Portapack crashes
see short video I recorded, happens also from other apps but only when going back to Audio

[IMG_5604.mp4.zip](https://github.com/furrtek/portapack-havoc/files/4558664/IMG_5604.mp4.zip)

i m using the version just downloaded and compiled 2-3 hours ago.
the version I downloaded and compiled about three weeks ago (04-08 I think) did not have this issue as far as I know

eried, 2020-05-02:
Can you test both versions and be sure that the older one works? If you tell me the timestamp of the one that works vs the one that does not I can check the differences.

BTW I compiled the "latest" -> https://github.com/eried/portapack-havoc/releases

helmarw, 2020-05-02:
k, will let you know, luckily i saved the "old" one, alt least the binary.
i will load it up and see if i can reproduce it there

helmarw, 2020-05-02:
my mistake, havnt teste the older verison (downloaded and compiled 8th April 2020)
has the same error. 
it only happens when Audio was in WFM-mode, switching to TPMS and than back to Audio crashes the whole thing
second when e put audio in SPEC-mode and then change to TPMS i get a red screen (M4 GURU Meditation unhandled)... two short videos about that behaviour
[crashes.zip](https://github.com/furrtek/portapack-havoc/files/4567960/crashes.zip)

other modes AM and NFM are not influenced either way. I didnt test any other apps, to many possible combination, it just happend that i was using exactly those two configurations for testing

eried, 2020-05-03:
Strange I do not get a lock if I repeat exactly what you do in those videos, it might be a faulty portapack?

helmarw, 2020-05-03:
i dont know what the Problem can be, that just how it is with my portapack

thnx, the housing is great, i printed it on a Formlabs Form2, work very well :)

fcanaan, 2020-05-06:
I can replicate this as well. If i go into the audio app, set it to WFM, exit and then go into ACARS or TPMS i get a had lock with a red screen reading "M4 Guru Medidtation Unhandled". I can also lock up the audio reciever, by going into WFM mode, exiting audio receiver, and then reopening it. If i can provide anymore information, please let me know.

eried, 2020-05-06:
I do not get the exact same type of lock up, but if I do audio, ACARS/TPMS and then back to audio, the audio app locks the portapack (only the upper part refreshes). Maybe some missing initialization stuff, I will play around it to see if I can find something

fcanaan, 2020-05-06:
That matches exactly one of my failure modes. the Guru
meditation doesnt happen every time for whatever reason. How maddening

eried, 2020-05-06:
Only with the audio app? I could check the initialization of that one but it would be helpful if you both @fcanaan @helmarw test **all** 👷‍♂️ the possible entering/exiting situations that work weird to pin point what is going on.

fcanaan, 2020-05-06:
Yes, i've only noticed it with the audio app, specifically in WFM mode. I
can help you test whatever scenarios you need.

eried, 2020-05-06:
Find the same problem in another app/in-out combination. Im no expert on the code, but with a few examples, it will be evident to fix 😂 I hope lol

fcanaan, 2020-05-06:
Strangely, the only error I can now replicate over and over is open audio,
switch to wfm mode, exit audio, open audio again, crash. Top 1/3 of
receiver menu shows the rest of the screen is black. Hard lock, have to
reset.

IF I open another Listener app, close it,and then open audio again, it does
not crash. Will keep testing.

fcanaan, 2020-05-07:
I’ve found a repeatable M4 Guru error. 
Turn unit on
Select Receivers
select pocsag wait 5 seconds then exit
Select audio change to spec mode wait 5 seconds then exit
select radiosonde
M4 Guru Meditation Unhandled hard lock up.

eried, 2020-05-07:
I can confirm tht M4 guru error, awesome! I will buzz around the code to see if I can find the source of the problem

eried, 2020-05-08:
I think I found a way to make it stop hanging :) it needs a call to `on_modulation_changed(` in the constructor

@fcanaan have you found more bugs? I want to be sure I am not breaking something else before posting the fix

eried, 2020-05-08:
@fcanaan @helmarw take a look and give me feedback of https://github.com/eried/portapack-havoc/releases/download/nightly/portapack-nightly_202005081310.bin It solves the 1/3 screen hang, but not the M4 guru error yet.

nemanjan00, 2020-05-08:
For me, it is hanging if I go to capture, and then, after that into audio reciever

eried, 2020-05-08:
> For me, it is hanging if I go to capture, and then, after that into audio reciever

OK you are right, this is a different kind of hang tho. We have 1/3 of screen, M4 guru meditation and "just frozen" now.

I think there is some stuff not cleanly terminated or such, because this https://github.com/furrtek/portapack-havoc/pull/353 is just calling the on_modulation_changed (which calls again the update_modulation ) and it fixes partially some issues. I am doing some test in pocsag and radiosonde constructors/destructors to see if I find the real issue.

eried, 2020-05-08:
OK! so I found the M4 guru meditation problem, SPEC just leaves the sampling rate too high... so this issue would be 100% solved...  "And I Would've Gotten Away With It Too. If It Weren't For You Meddling @nemanjan00 ." xDDDDD

eried, 2020-05-08:
OK @nemanjan00 @fcanaan @helmarw 

Try to hang: https://github.com/eried/portapack-havoc/releases/download/nightly/portapack-nightly_202005061640.bin somehow. 

I think this ticket could be closed.

nemanjan00, 2020-05-08:
@eried, I did not test that image, but, image built from your master does seem to work without any crashes for me.

fcanaan, 2020-05-08:
I'll toss the image on my unit at lunchtime and report back to you.

fcanaan, 2020-05-09:
Loaded bin. Looking good. Haven't been able to crash it yet. Will keep trying. Thanks!

fcanaan, 2020-05-14:
No crashes in the image yet. I have found that the map does not load when you view a flight picked up by the ads-b app. The worldmap.bin has been generated correctly and is in the correct spot on the sd card. I even reverted to the last public image I had, and the map functionality came back.

eried, 2020-05-14:
@fcanaan yes I changed the map file for a mercator one instead of equidistant, you can download my [world_map.bin](https://github.com/eried/portapack-mayhem/releases) from sdcard.rar or check the pullrequests here to create yours from the updated jpg

helmarw, 2020-05-20:
hi, sorry had to take care of other stuff last few weeks. great to see the development, I will try out the updated firmware soon!

helmarw, 2020-05-20:
> my mistake, havnt tested the older version very much (downloaded and compiled 8th April 2020). it has the same error.
> it only happens when Audio is in WFM-mode, switching to TPMS and than back to Audio. that crashes the whole thing
> second when i put audio in SPEC-mode and then change to TPMS i get a red screen (M4 GURU Meditation unhandled)... two short videos about that behaviour
> [crashes.zip](https://github.com/furrtek/portapack-havoc/files/4567960/crashes.zip)
> 
> other modes AM and NFM are not influenced either way. I didnt test any other apps, too many possible combination, it just happend that i was using exactly those two configurations for testing my TPMS sensors

GURU still there but Audio doesnt crash anymore, at least not that I could make it

eried, 2020-05-20:
I cannot trigger guru anymore, how are you triggering it?

helmarw, 2020-05-20:
im going into Audio and displaying SPEC and then exit Audio and start TPMS

eried, 2020-05-20:
Nope, I fixed that :) it was merged today here or you can check my release bin too

helmarw, 2020-05-20:
i recompiled it a aprox 11 hours ago from a fresh git clone
or did you commit something afterwards?

eried, 2020-05-20:
The PR was accepted like 9 hours ago https://github.com/furrtek/portapack-havoc/pull/356, so if you compiled this one 11 hr ago, you still have the guru m4 error

helmarw, 2020-05-20:
k makes sense, i was to quick.
i used your nightly built now, and its working 
thnx!!! 👍