furrtek/portapack-havoc #358: lpc43xx_cpp.hpp error

decadenzio, 2020-05-11:
The build stops at 12%. Followed the wiki, installed all dependencies.

`[ 12%] Building CXX object firmware/baseband/CMakeFiles/baseband_shared.dir/baseband.cpp.obj
In file included from /usr/include/newlib/sys/reent.h:503,
                 from /usr/include/newlib/stdlib.h:18,
                 from /usr/include/newlib/c++/8.3.1/cstdlib:75,
                 from /usr/include/newlib/c++/8.3.1/bits/stl_algo.h:59,
                 from /usr/include/newlib/c++/8.3.1/algorithm:62,
                 from /root/portapack-havoc/firmware/common/utility.hpp:28,
                 from /root/portapack-havoc/firmware/common/lpc43xx_cpp.hpp:29,
                 from /root/portapack-havoc/firmware/baseband/baseband.cpp:24:
/root/portapack-havoc/firmware/common/lpc43xx_cpp.hpp:72:13: error: expected unqualified-id before ')' token
 inline void assert() {
             ^~~~~~
/root/portapack-havoc/firmware/common/lpc43xx_cpp.hpp:72:13: error: expected ')' before '?' token
 inline void assert() {
             ^~~~~~
In file included from /root/portapack-havoc/firmware/common/message.hpp:36,
                 from /root/portapack-havoc/firmware/common/message_queue.hpp:27,
                 from /root/portapack-havoc/firmware/common/portapack_shared_memory.hpp:29,
                 from /root/portapack-havoc/firmware/baseband/baseband.cpp:26:
/root/portapack-havoc/firmware/common/adsb_frame.hpp: In member function 'uint8_t* adsb::ADSBFrame::get_raw_data() const':
/root/portapack-havoc/firmware/common/adsb_frame.hpp:68:26: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
   return (uint8_t* const)raw_data;
                          ^~~~~~~~
make[3]: *** [firmware/baseband/CMakeFiles/baseband_shared.dir/build.make:908: firmware/baseband/CMakeFiles/baseband_shared.dir/baseband.cpp.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:1497: firmware/baseband/CMakeFiles/baseband_shared.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:691: firmware/CMakeFiles/firmware.dir/rule] Error 2
make: *** [Makefile:326: firmware] Error 2
`

z4ziggy, 2020-05-14:
Were you able to fix this? The only way I could avoid this for now is renaming the assert() to xassert() (or whatever), including all the caller-functions ofc. Didn't have time to investigate this further.

z4ziggy, 2020-05-14:
You can also comment out "_REENT_CHECK_VERIFY" in /usr/include/newlib/nano/newlib.h

I couldn't find a better fix for now.

z4ziggy, 2020-05-14:
and here is the correct patch:
```
diff --git a/firmware/common/lpc43xx_cpp.hpp b/firmware/common/lpc43xx_cpp.hpp
index 13ca9aa7..1ca37b98 100644
--- a/firmware/common/lpc43xx_cpp.hpp
+++ b/firmware/common/lpc43xx_cpp.hpp
@@ -69,6 +69,7 @@ inline void disable() {
 #endif
 
 #if defined(LPC43XX_M4)
+#undef assert
 inline void assert() {
        __SEV();
 }
@@ -93,6 +94,7 @@ inline void disable() {
 #endif
 
 #if defined(LPC43XX_M0)
+#undef assert
 inline void assert() {
        __SEV();
 }

```

decadenzio, 2020-05-14:
so it's missing the line `#undef assert` before the two `inline void assert ()` declarations.
I will try modifying the file, compile and post an update. 

Against which other `lpc433xx_cpp.hpp` version did you run `diff`?

decadenzio, 2020-05-14:
It actually compiles it without problems, just a few minor warnings.
I will close the thread, thanks!

closed 2020-05-14 by decadenzio