furrtek/portapack-havoc #361: I would like to have more than 99 lines in /FREQMAN files

alain00091, 2020-05-20:
it is right now limited by the code  in freqman.hpp " #define FREQMAN_MAX_PER_FILE 99" , I changed it to 499 and recompile the code
Scanner is working perfectly but File manager and frequency manager crashes

An example of file is 240 channels of CB (lower/superior channels) 

I just started to learn C++, sorry for that

furrtek, 2020-05-20:
That's why FREQMAN_MAX_PER_FILE is defined to 99 and not 499 :)
There isn't much RAM to play with.

alain00091, 2020-05-20:
Yes I understand your answer but for robustness firmware should check if there is enough memory before mem allocation

furrtek, 2020-06-16:
There's no dynamic memory allocation.
You changed a constant without checking how it was used.

closed 2020-06-16 by furrtek

alain00091, 2020-06-16:
Yes I know it now after I read the code
I made a lot of test with many files and huge files, so it is always possible to crash it even if we use the magic number 99 ...
A better way is to use old style malloc/calloc dynamic allocation, I need to go in deep in the code to check vector allocation, it should be dynamic