furrtek/portapack-havoc #363: scanner improvement ( NFM + AM + ...)

alain00091, 2020-05-20:
Is it possible to add a new demodulation to the scanner

1 maybe create 2 files  "SCANNER_AM"  and "SCANNER_NFM" and select which one we want to scan or

2 in the SCANNER file add at the end add the demodulation method like AM.NFM.SSB,, etc ... or
3 when we start the scanner ask which demodulation to use

eried, 2020-05-20:
They can be probably added, but by whom? Since you already explored the code, maybe you can start poking in the scanner app code. I support the idea of those new options 😁

alain00091, 2020-05-20:
Thanks Eried, the problem is that I just started learning C++ with furtek/mayhem firmware, so if I do it it will take a while... but I will try ( I'm retired ...) !